Qucs/qucsator #5: harmonic balance error

michal777, 2020-09-29:
Hi
I tried simple harmonic balance simulation, I get an error:
`terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 16) >= this->size() (which is 16)`
[image]

JSebastianOspina, 2021-03-21:
Could anyone fix this ? Im getting same error in the last qucs version

felix-salfelder, 2021-03-21:
On Sat, Mar 20, 2021 at 05:32:38PM -0700, Sebastian Ospina wrote:
> Could anyone fix this ? Im getting same error in the last qucs version

Not sure who is going to fix this. Few people working on Qucsator these
days.

Please could you share a text representation of the schematic and
netlist.txt just in case?