Qucs/qucsator #24: Voltage probe handles noise voltages incorrectly

tejeez, 2021-02-25:
In an AC simulation, the voltage probe appears to simply calculate the difference in total noise voltage between its terminals, which generally does not provide the correct result.

As a simple demonstration of the problem, let's try measuring the voltage between two uncorrelated noise sources of 1 V^2/Hz:
[image]

The resulting Pr1.vn is 0, whereas I think the correct result would be √2 V/√Hz.

Now, if I replace the voltage probe with an ideal transformer and a label to perform the same measurement, I get the correct result:
[image]

The "transformer and a label" trick can be used a workaround for now. It seems to be a bug in the implementation of the voltage probe itself.

I'm running the latest Qucs version compiled from develop branch 29432325df75d0c1512624a2c5811e99e4342058 and Qucsator 0.0.20, a08b0a146505d926c6cc966113450006e974af79 . (Btw, I'm not sure whether this issue should in the qucs or in the qucsator repository).

felix-salfelder, 2021-02-27:
Thanks for the bug report.

> (Btw, I'm not sure whether
> this issue should in the qucs or in the qucsator repository).

That will require some investigation, either way.

e.g. Could you have a look at the Qucsator output? Is it correct?
How does the voltage probe handle noise voltages?

ckoegler, 2021-07-18:
qucsator output:
<Qucs Dataset 0.0.20>
<indep acfrequency 2>
  +1.00000000000000000000e+06
  +2.00000000000000000000e+06
</indep>
<dep Pr1.v acfrequency>
  +0.00000000000000000000e+00
  +0.00000000000000000000e+00
</dep>
<dep _net0.v acfrequency>
  +0.00000000000000000000e+00
  +0.00000000000000000000e+00
</dep>
<dep _net1.v acfrequency>
  +0.00000000000000000000e+00
  +0.00000000000000000000e+00
</dep>

looks identical to the tabular output in the schematic

[vol_probe_noise.sch.txt](https://github.com/Qucs/qucsator/files/6836867/vol_probe_noise.sch.txt)