Qucs/qucsator #26: SPfile file interpolation problem

thliebig, 2021-03-16:
If the phase in the touchstone file is random or changing quickly the frequency interpolation produces false data.
I assume the interpolation is done base on the real/imaginary values, but I think it should be done based on amp/phase instead.
Ideally it should be done base on the specified format (MA, DB or RI). Any comments on this assumption are welcome. 

A test file is attached. If the simulation is setup with 0 to 10GHz and 101 points all is good, but e.g. if the number of points is set too 100 or any other number the results start changing drastically.

Can someone point to me to a file or function to look at for the interpolation? Any ideas how to fix this?

[test.s1p.zip](https://github.com/Qucs/qucsator/files/6148063/test.s1p.zip)

guitorri, 2021-03-16:
Indeed, it seems to do real/imaginary.

I guess you are looking for:
- https://github.com/Qucs/qucsator/blob/develop/src/components/spfile.cpp
- https://github.com/Qucs/qucsator/blob/develop/src/interpolator.cpp

thliebig, 2021-03-16:
Thanks for the quick respone and yes I found this files too. Thank you!
I think changing the line 68 in [spfile](https://github.com/Qucs/qucsator/blob/develop/src/components/spfile.cpp) does the trick:
from:
`inter->prepare (it, REPEAT_NO, dt | DATA_COMPLEX);`
to:
`inter->prepare (it, REPEAT_NO, dt | DATA_COMPLEX | DATA_POLAR);`
Do you think this is correct? Shall I prepare a PR?

thliebig, 2021-03-16:
Okay I think we do not need to change anything. The SPFile can be set to polar (instead of "rectangular") in the settings and all looks good!!
That means it just looks like a user error? Maybe the default could/should be changed?
Feel free to close, I think no changes are needed...

closed 2021-03-16 by thliebig