Qucs/qucsator #29: Simulator crashes with glibc 2.33

Lacsapix, 2021-06-08:
Trying to simulate a simple voltage divider, the simulation crashes with the following helpfull message: "ERROR: Simulator crashed!"

Digging a bit around I did: 

    qucsator -i netlist.txt

And it provided a bit more help, a coredump:

    /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/10.2.0/../../../../include/c++/10.2.0/bits/stl_vector.h:1045: std::vector::reference std::vector<qucs::nodelist_t *, std::allocator<qucs::nodelist_t *>>::operator[](std::vector::size_type) [_Tp = qucs::nodelist_t *, _Alloc = std::allocator<qucs::nodelist_t *>]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
    Aborted (core dumped)

Researching  a bit more: I found that the following code provides the same error [https://github.com/rstudio/httpuv/issues/133#issuecomment-387499519](url)

This led me to believe that glibc might be the culprit here. Surely enough when browsing through the Archlinux forums My suspicion was confirmed by the (current) top post on the AUR page of QUCS [https://aur.archlinux.org/packages/qucs/](url). It seems the upgrade from glibc 2.25 to version 2.xx throws this error. Running the same simulation on a different computer (also an archlinux pc) with hasn't upgraded in months, seems to work fine.

Of course its Archlinux which is known for having crashes when upgrading packages to soon. But since more Distro's are likely to follow, it will come up more often.

[netlist.txt](https://github.com/Qucs/qucs/files/6614961/netlist.txt)

Lacsapix, 2021-06-09:
Yes I've found that flag as well.
Another note to add: I've upgraded my "different computer" to the latest archlinux repo's (with glibc 2.33-5) and qucs still works...... So this a bit more strange than I initially thought. Maybe another missing dependency.

felix-salfelder, 2021-07-16:
I tried on Debian with g++-{9,10,11}. A production build does not crash (maybe this is where the glibc version comes in), but a build with -D_GLIBCXX_ASSERTIONS exposes the bug.

In particular, I can now test/merge a patch if you send one.

ckoegler, 2021-07-18:
I am working on this.

ckoegler, 2021-07-21:
The problem was the write access to std vector narray 
narray[0] = n;
and
narray[i] = n;
in function nodelist::assignNodes before reserving this memory properly.

felix-salfelder, 2021-08-04:
On Wed, Aug 04, 2021 at 02:15:32AM -0700, Alessandro Paganelli wrote:
> Confirmed also on Fedora 34 with:
> - Qucs 0.0.19
> - Qucsator 0.0.19
> - glibc 2.33-20-fc34

Thanks for confirmation. It has been fixed (partially) with #30, merged
into develop.

The crash is not a big surprise when you look at the code. But we are
not sure if there might be more instances of this. Please test the fixed
version.

Please check if you could use Qucsator 0.0.20 instead, on Fedora.

allep, 2021-08-04:
A newer package is not available on Fedora 34. I rebuilt it from tag 0.0.20 as you suggested and now everything works fine. Thanks!

felix-salfelder, 2021-08-04:
Thanks. Please open a new issue if there's more of this.

closed 2021-08-04 by felix-salfelder