Qucs/qucsator #34: Frequency dispersion of impedance and dielectric constant is not used to compute the losses of microstripline

mhuser, 2022-05-10:
While the frequency dispersion of impedance and dielectric constant of the microstripline are computed, they are not used to compute the losses.
https://github.com/Qucs/qucsator/blob/22126bb9efe0f3d91d2c772cebde3f7ff1b619ef/src/components/microstrip/msline.cpp#L75-L80
However they are properly used to compute gamma.
https://github.com/Qucs/qucsator/blob/22126bb9efe0f3d91d2c772cebde3f7ff1b619ef/src/components/microstrip/msline.cpp#L82-L86
The original paper of Hammerstad and Jensen is ambiguous about this, as it used the same variable names for these values with or without dispersion.
On the other hand, it sounds strange to use the quasi-static constant value for losses and a frequency-dependant value for gamma. This eventually cause Qucs to have a small difference in simulated insertion loss if compared against Keysight ADS (using frequency invariant dielectric and kirschning models).

felix-salfelder, 2022-05-11:
Thanks for reporting this.

On Tue, May 10, 2022 at 10:42:10AM -0700, mhuser wrote:
> While the frequency dispersion of impedance and dielectric constant of
> the microstripline are computed, they are not used to compute the
> losses.

I think you best bet is so send a patch. It does sound like an
oversight in the current model.

> This eventually cause Qucs to have a small difference in simulated
> insertion loss if compared against Keysight ADS (using frequency
> invariant dielectric and kirschning models).

Will the difference completely disappear with your changes? This would
be a strong point -- perhaps there is no need to keep a wrong version.

mhuser, 2022-05-12:
You are welcome. I will prepare some plots to illustrate the behaviour with/without the proposed modification.

mhuser, 2022-05-14:
First let me introduce the context: I am contributing to [scikit-rf](https://github.com/scikit-rf/scikit-rf) a Python package that provide standard microwave network operations, such as reading/writing Touchstone files (.sNp files), connecting or de-embedding N-port networks, .... and advanced operations such as communication with Vector Network Analyzer (VNA) amongst other things that are interesting to automate in scripts without a graphical interface.

Of course, we are not rewriting this great tool that is QUCS in Python, instead [we are using it](https://scikit-rf.readthedocs.io/en/latest/examples/networktheory/IEEEP370%20Deembedding.html#id1) to build networks out of its extensive component library and process later with the resulting s-parameters files.

At a point, it appeared it was useful to implement the msline model into the package to use Python optimizations to fit microstripline models on measurement, e.g. to determine relative permittivity of substrates. We inspired highly of qucsator [qucs msline](https://github.com/Qucs/qucsator/blob/develop/src/components/microstrip/msline.cpp) implementation because it is open and the equations are very-well documented.

We checked that our implementation follows qucsator, but during the process we had a chance to compare to [ads mlin](https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5693243) which is a closed commercial solution. We noticed a discrepancy in the insertion loss results and after looking at it, using the frequency-dispersed values for the loss computation significantly enhance the agreement. The original paper use the same variables names for quasi-static or frequency-dispersed values so there is an ambiguity, but it make sense to use frequency-dependant values here (they are used for propagation, which will also embed the losses).

[image]

[image]

mhuser, 2022-05-14:
This is our issue for reference:
https://github.com/scikit-rf/scikit-rf/issues/647

There is also a difference between QUCS and ADS on the Return Loss. But ADS implementation is questionable there as they use a complex permittivity (ep_r - 1j ep_r * tand), which lead to characteristic of the line having a complex value. I think this is done to use their dielectric frequency-dispersion model (Svensson/Djordjecic wideband Debye) implementation.