Qucs/qucsator #36: random() is crashing the simulator

Wallacegromit, 2023-01-16:


felix-salfelder, 2023-01-16:
On Mon, Jan 16, 2023 at 05:11:30AM -0800, Wallacegromit wrote:
> # Qucs 0.0.20  /pro/share/qucs/ex_stat_prj/verification_random_values.sch
> 
> Eqn:Eqn2 var_x="1" Export="yes"
> Eqn:Eqn3 value="random()" Export="yes"
> Eqn:Eqn1 seed="srandom(1)" Export="yes"
> .DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
> .SW:SW1 Sim="DC1" Type="lin" Param="init" Start="1" Stop="10" Points="10"

Crashes all the same for me, please fix it if you can.

The closest I got to random numbers in Qucs simulations is with Gnucsator.

input "netlist.txt":

Idc:I1 _net0 gnd I=".1"
.SW:SW1 Sim="DC1" Type="lin" Param="swp" Start="10m" Stop="50m" Points="5"
Eqn:Eqn3 value="aunif(0,1)" Export="yes"
Eqn:Eqn1 swp="1" Export="yes"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
R:R1 _net0 gnd R="value"

output "dat file":

<indep swp 5>
 0.01
 0.02
 0.03
 0.04
 0.05
</indep>
<dep _net0.V swp>
-0.05736191
 0.08874336
-277.938E-6
-0.07651821
 0.02742623
</dep>

All you need is "load path/to/f_random.so" from [1] in gnucsator.rc or
added at the top of your netlist.txt. Perhaps f_random.cc (or a variant
of it) should be shipped with Gnucsator, feel free to send a patch or
discuss further.

https://git.savannah.gnu.org/cgit/gnucap/gnucap-plugins.git/log/?h=random

Wallacegromit, 2023-01-17:
knowing that **it was working before on previous version**. I have used it to trace stats on filters using the precision of the components. I took values of components Cap = Co * ( 1 + precision (random()-0.5+ nbIter * 0 ) . doing a parameter sweep on nbiter ( 1 to n )  n:is the amount of samples I want. I was able to plot the cutoff frequencz variation due to component precision.

felix-salfelder, 2023-01-17:
On Mon, Jan 16, 2023 at 11:44:17PM -0800, Wallacegromit wrote:
> knowing that **it was working before on previous version**.

Look at the node allocation bug (a.k.a. "crash with newer glibc" or
so.). This one might be similar.