Qucs/qucsator #37: Assertion in stl_vector.h:1123 on Fedora 28

KavalchukD, 2023-05-16:


KavalchukD, 2023-05-16:
Hey Felix!
Sure. I should admit I've faced some issues while building the project locally. I had decided to use cmake + ninja, and qucsator file didn't seem to run. It hung for at least several minutes as far as I checked.

KavalchukD, 2023-05-16:
Running bootstrap to generate configure causes 
sh bootstrap
bootstrapping the qucs-core sources...
Creating aclocal.m4... bootstrap: line 33: aclocal: command not found
Here are those lines:
echo -n "Creating aclocal.m4... "
${ACLOCAL:-aclocal} -I m4 || exit 1

KavalchukD, 2023-05-16:
It seems I failed to recognise the structure of this project. I built a correct project which is of course qucs itself, and the object of the issue is seems fixed on the actual revision. 
[image]

closed 2023-05-16 by KavalchukD

schnittchen, 2025-02-16:
Nice to learn that the issue I just bumped into as a new user seems to be fixed, but I'm not going to build a QT Project from sources...

felix-salfelder, 2025-02-16:
On Sun, Feb 16, 2025 at 05:37:27AM -0800, Schnittchen wrote:
> Nice to learn that the issue I just bumped into as a new user seems to
> be fixed,

You are welcome.

There hasn't been much interest in Qucsator lately. For further
development or a release, we'd need to find a maintainer.

Not much else to see here. If you only need binaries for Fedora, ask
them to supply a patched version.

> but I'm not going to build a QT Project from sources...

What makes you believe that Qucsator is a "QT Project"? If there is an
error in the documentation, please point it out in a new ticket.