Qucs/qucsator #41: Compile error... missing scan_citi.cpp

Rmano, 2023-12-14:
After running bootstrap and configure correctly, I have this error: (`develop` branch):

```
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I../src/math -I../src/components -I../src/components -I../src/interface -I. -O2 -pipe -fno-exceptions -fno-check-new -ldl -rdynamic -MT scan_citi.lo -MD -MP -MF .deps/scan_citi.Tpo -c scan_citi.cpp  -fPIC -DPIC -o .libs/scan_citi.o
cc1plus: fatal error: scan_citi.cpp: No such file or directory
```

Any hint/help?

felix-salfelder, 2023-12-14:
On Thu, Dec 14, 2023 at 05:34:30AM -0800, Romano Giannetti wrote:
> Any hint/help?

Could be a makefile bug or some lex/yacc issue? Here, I get

src$ rm scan_citi.cpp
src$ make scan_citi.cpp
/bin/bash ../../build-aux/ylwrap ../../src/scan_citi.lpp lex.citi_.c scan_citi.cpp -- flex --nounistd

Hope that helps.

cheers
felix

Rmano, 2023-12-14:
Ok, I had no `flex` installed. Strange, I supposed that the configure script should have noticed that...

closed 2023-12-14 by Rmano

felix-salfelder, 2023-12-14:
On Thu, Dec 14, 2023 at 08:21:34AM -0800, Romano Giannetti wrote:
> Ok, I had no `flex` installed. Strange, I supposed that the configure script should have noticed that...

Might be just a one-liner in configure.ac.

Feel free to send a patch..

cheers
felix

Rmano, 2023-12-14:
> On Thu, Dec 14, 2023 at 08:21:34AM -0800, Romano Giannetti wrote: Ok, I had no `flex` installed. Strange, 

> Might be just a one-liner in configure.ac. Feel free to send a patch.. 

Uf, that file is like `perl` to me.... ;-P (meaning, no way to understand a thing!).

BTW, I think that having `qucsator`, which now works flawlessly with the `qucs-s` interface, is a great thing. No need to repeat the work of playing catch with qt releases in two different projects... ;-) 

Thanks again.