Qucs/qucsator #42: S-parameter file causes simulation error

zodoczi, 2024-03-02:
Hi, 

I'm playing with this RF switch, where quite hopeless what should be wrong with my simulation as many different switches was working before. I copy the simulation error log also attach the S parameter file here. In a text editor I was not finding any strange things however sure there is something not usual here. Thats sure that the S parameter file was made with a real desktop VNA.

`Errors and Warnings:
--------------------

line 23: no trailing end-of-line found, continuing...
line 23: syntax error, unexpected Float, expecting $end
checker error, found 6 options
checker error, option `hz' occurred 2x
checker error, option `s' occurred 2x
checker error, option `ri' occurred 2x
checker error, option `hz' occurred 2x
checker error, option `s' occurred 2x
checker error, option `ri' occurred 2x
[D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A.s5p.zip](https://github.com/Qucs/qucs/files/14460361/D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A.s5p.zip)
`

Thank you very much!

zodoczi, 2024-03-02:
Sure please go ahead and use this to solve the case! Thanks

What I see in the MWO fixer tool is only this bug: "Negative character used as a delimiter in file; fixed"

felix-salfelder, 2024-03-02:
On Sat, Mar 02, 2024 at 06:31:58AM -0800, Zoltan Doczi wrote:
> Sure please go ahead and use this to solve the case! Thanks
> 
> What I see in the MWO fixer tool is only this bug: "Negative character used as a delimiter in file; fixed"

Perhaps MWO does not consider long lines a problem, but only dumps short
lines as a measure of precaution. Or to be compliant with some
touchstone spec...

The qucs touchstone reader uses lex/yacc, and the grammar seems to limit
the number of Float tokens per line. Rookie mistake, either in the spec,
or in the implementation.

Anyway, I won't use lex or yacc, and this is going to be a non-issue.

felix-salfelder, 2024-03-18:
[gnucsator](https://codeberg.org/gnucap/gnucsator) now has an SPFile component that works with your original s5p (same numerical result as qucsator, but following a different approach).

Thanks for your example and test case. That's the kind of stuff I need...

(Curiously qucsator/SPfile transposes the S param matrix if there are only two ports. I wonder if this is intentional.)