Qucs/qucsator #48: S-parameter max size or memory size limitation

buenoshun, 2024-07-08:
There was a topic already on sourceforge, this mirrors it here.
https://sourceforge.net/p/qucs/discussion/311049/thread/ded7d8c9/

Either the S-parameter file parser or the qucs simulator has a memory or file size limit.
If I try to import anything larger than maybe 60MBytes S-parameter file, the simulator crashes with an error.
For professional simulations in digital electronics, we need to chain together multiple S-par files that are 32 ports each, having maybe 4000 samples, that is more than 500MBytes. If we insert 4 Spar files, each being 600MBytes, that is 2.4GBytes.

Typical error message:
line 500880: memory exhausted
checker error, found 6 options
checker error, option `s' occurred 2x
checker error, option `s' occurred 2x

When releasing a new version of QUCS, please increase this limitation, to allow up to 1GBytes S-parameter files each, or 3GBytes or more total for all S-par files in the same schematic combined.

felix-salfelder, 2024-07-08:
Moved to Qucsator. looks similar to #42.

felix-salfelder, 2024-07-08:
#42 is about format, this one is about size.