Qucs/qucsator #50: Lack of Lambert W function in Qucsator

domi8space, 2024-10-23:
Hi all,

I am new to GitHub, which i join to help improving Qucs-based softwares (including Qucs-S), and this is my first post.

A famous mathematical function, often found when it is needed to solve equations with a linear term and either a logarithmic or exponential term, is lacking in the Qucs family's set of mathematical functions: the Lambert's W function.
https://en.wikipedia.org/wiki/Lambert_W_function

There are some papers which give some examples of such electronic circuits, and even Wikipedia mentions the "diode and resistor" circuit solving with this function.

Most numerical computation softwares (Maple, Mathematica, Matlab, Scilab, Octave, Python, etc.) have this feature. Because this function can solve many problems that can be found in electronic circuits involving diodes or bipolar transistors, I think it would be useful to see it in Qucsator-based simulation softwares. I personally have to export the simulation data from Qucs(-S) and use an external software to plot the simulated results with the theoretical ones that use the Lambert W function, which makes it painful for repetitive iterations.

Thanks!

felix-salfelder, 2024-10-24:
On Wed, Oct 23, 2024 at 09:33:14AM -0700, domi8space wrote:
> I am new to GitHub, which i join to help improving Qucs-based
> softwares (including Qucs-S), and this is my first post.

(Sorry about Github. Many would prefer keeping the stuff together. E.g.
on the mailing list. But switching off the message boards here might
leave us with even less ...)

> https://en.wikipedia.org/wiki/Lambert_W_function

Thanks for sharing. Can you give a quick explanation how you would use
the function? Do you intend to use it in device modelling or in
postprocessing? Do you need real or also complex arguments? How about
the branch selection? Please provide an example use, e.g. in terms of a
Qucsator netlist?

> Most numerical computation softwares (Maple, Mathematica, Matlab, Scilab, Octave, Python, etc.)

I found a C++ implementation in scipy under a suitable license.
According to a comment in the code, it has an issue "close to the branch
point". It uses some of the scipy infrastructure. But it might serve as
a starting point.

ra3xdh, 2024-10-24:
> I am new to GitHub, which I join to help improving Qucs-based softwares (including Qucs-S)

This repository is not related to Qucs-S. The issues and proposals for Qucs-S should be reported here: 
https://github.com/ra3xdh/qucs_s/issues
https://github.com/ra3xdh/qucs_s/discussions
https://github.com/ra3xdh/qucsator_rf/issues

> Because this function can solve many problems that can be found in electronic circuits involving diodes or bipolar transistors

I need more info how this is used for circuit modelling. Usually the resistor connected in series to p-n junction is sufficient and explicit Lambert function is not required. I cannot remember if proprietary circuit simulation software (Spectre, HSPICE) have the support of Lambert function.

domi8space, 2024-10-25:
Thanks for your replies. 

Let me add a few more useful materials to my previous post's mention of existing research about the Lambert W function in the analog electronics domain:
- "Recent Applications of Lambert's W Function in Nanodevice Modeling Introduction", 2024, https://www.researchgate.net/publication/380637830_Recent_Applications_of_Lambert%27s_W_Function_in_Nanodevice_Modeling_Introduction (open access)
- "Bipolar transistor circuit analysis using the Lambert W-function", 2002, https://ieeexplore.ieee.org/document/895330 (restricted access)

I personally think to use such a function for postprocessing (to plot a theoretical quantity along with the same quantity simulated). However I do not exclude that other people may find it useful for device modelling. 

As far as I know, only real arguments are needed. This is the case for the few examples I know. 

Please find below the code of a Qucs schematic using Qucsator (Qucs-S 1.1.0). This case uses only the W0 branch with w*exp(w)>=0, but there exist circuits that use both branches (W0 and W-1) with w*exp(w)<0.

<Qucs Schematic 1.1.0>
<Properties>
  <View=-20,-560,1420,742,1,0,180>
  <Grid=10,10,1>
  <DataSet=Lambert_W_Function_Widlar_for_GitHub.dat>
  <DataDisplay=Lambert_W_Function_Widlar_for_GitHub.dpl>
  <OpenDisplay=0>
  <Script=Lambert_W_Function_Widlar_for_GitHub.m>
  <RunScript=0>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
</Symbol>
<Components>
  <.DC DC1 1 120 -330 0 40 0 0 "26.85" 0 "0.001" 0 "1 pA" 0 "1 uV" 0 "no" 0 "150" 0 "no" 0 "none" 0 "CroutLU" 0>
  <Eqn Eqn1 1 870 10 -30 16 0 0 "u_T=vt(300)" 1 "Iref=w_ew*u_T/R3" 1 "yes" 0>
  <Eqn Eqn2 1 870 -70 -30 16 0 0 "R3=1 k" 1 "yes" 0>
  <Eqn Eqn5 1 1110 -50 -30 16 0 0 "Beta=200" 1 "yes" 0>
  <.SW SW1 1 290 -330 0 65 0 0 "DC1" 1 "lin" 1 "w_ew" 1 "0" 1 "6" 1 "601" 1 "false" 0>
  <GND * 1 410 210 0 0 0 0>
  <_BJT T1 1 260 70 -81 -26 1 2 "npn" 1 "1e-16" 1 "1" 1 "1" 0 "0" 0 "0" 0 "0" 1 "0" 0 "0" 0 "1.5" 0 "0" 0 "2" 0 "Beta" 1 "1" 0 "0" 0 "0" 0 "0" 0 "0" 0 "0" 0 "0" 0 "0.75" 0 "0.33" 0 "0" 0 "0.75" 0 "0.33" 0 "1.0" 0 "0" 0 "0.75" 0 "0" 0 "0.5" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "26.85" 0 "0.0" 0 "1.0" 0 "1.0" 0 "0.0" 0 "1.0" 0 "1.0" 0 "0.0" 0 "0.0" 0 "3.0" 0 "1.11" 0 "26.85" 0 "1.0" 0>
  <_BJT T2 1 410 70 8 -26 0 0 "npn" 1 "1e-16" 1 "1" 1 "1" 0 "0" 0 "0" 0 "0" 1 "0" 0 "0" 0 "1.5" 0 "0" 0 "2" 0 "Beta" 1 "1" 0 "0" 0 "0" 0 "0" 0 "0" 0 "0" 0 "0" 0 "0.75" 0 "0.33" 0 "0" 0 "0.75" 0 "0.33" 0 "1.0" 0 "0" 0 "0.75" 0 "0" 0 "0.5" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "0.0" 0 "26.85" 0 "0.0" 0 "1.0" 0 "1.0" 0 "0.0" 0 "1.0" 0 "1.0" 0 "0.0" 0 "0.0" 0 "3.0" 0 "1.11" 0 "26.85" 0 "1.0" 0>
  <R R1 1 410 180 15 -26 0 1 "R3" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "US" 0>
  <GND * 1 260 210 0 0 0 0>
  <Idc I1 1 260 -10 -74 -26 0 3 "Iref" 1>
  <Vdc V1 1 140 -70 18 -26 0 1 "10 V" 1>
  <GND * 1 140 -40 0 0 0 0>
  <IProbe Iout_sim 1 410 -30 38 -7 0 3>
  <Eqn Eqn3 1 870 130 -30 16 0 0 "w_sim=R3*Iout_sim.I/u_T" 1 "yes" 0>
</Components>
<Wires>
  <290 40 290 70 "" 0 0 0 "">
  <260 40 290 40 "" 0 0 0 "">
  <290 70 380 70 "" 0 0 0 "">
  <410 100 410 150 "" 0 0 0 "">
  <260 100 260 210 "" 0 0 0 "">
  <410 0 410 40 "" 0 0 0 "">
  <260 20 260 40 "" 0 0 0 "">
  <410 -110 410 -60 "" 0 0 0 "">
  <260 -110 260 -40 "" 0 0 0 "">
  <140 -110 140 -100 "" 0 0 0 "">
  <140 -110 260 -110 "" 0 0 0 "">
  <260 -110 410 -110 "" 0 0 0 "">
</Wires>
<Diagrams>
  <Rect 560 140 240 160 3 #c0c0c0 1 00 0 -1 1 6 0 0 0.5 2 1 -1 1 1 315 0 225 0 0 0 "" "" "">
	<"w_sim" #0000ff 2 3 0 0 0>
	  <Mkr 2.75 60 -224 5 0 0>
	  <Mkr 0 8 -150 3 0 0>
  </Rect>
  <Tab 1080 371 229 301 3 #c0c0c0 1 00 1 0 1 1 1 0 1 1 1 0 1 601 315 0 225 0 0 0 "" "" "">
	<"Iref" #0000ff 0 3 1 0 0>
	<"w_sim" #0000ff 0 3 1 0 0>
  </Tab>
</Diagrams>
<Paintings>
  <Arrow 490 -240 -100 70 20 8 #000000 2 1 1>
  <Text 500 -280 12 #000000 0 "w*ew is swept from 0 to 6. \nThe lower limit is set to 0 because this circuit does not \nhold the Lambert W function for e*ew<0 (I_{ref} \ncannot be negative).">
  <Text 280 -540 12 #000000 0 "Example of application of the Lambert W function with an \nanalog electronic circuit: the Widlar current source.\n\nThis circuit uses only the W0 branch with e*ew>=0. \nThere are other circuits whose electrical quantities depend on the Lambert W function. \nThe most simple is a diode in series with a resistor. \nThere are other circuits, with more complicated expressions, including some \nthat would use the branch W-1 (w*ew<0) as well as the branch W0 with w*ew<0 too. \n\ndomi8space, 25/10/2024">
  <Text 130 270 12 #000000 0 "The reference current is imposed. \nNeglecting base currents, the output current is given by: \nI_{out}= (u_T / R3) * W( (Is2/Is1) * R3 I_{ref} / u_T )\n\nor I_{out}= (u_T / R3) * W( x )\nwith x=(Is2/Is1) * R3 I_{ref} / u_T \nwhere W( x )=w \n\n(w is the solution to the equation with the form y=w*exp(w)">
  <Arrow 770 380 -30 -200 20 8 #000000 2 1 1>
  <Text 750 390 12 #000000 0 "This plot of the simulated w can be checked against the actual plot of \nthe Lambert W function (e.g.in Wikipedia). \nFor example, for w*ew=exp(1)=~2.71, one should found w_{sim}=~1. \nHere we find 2.75, mostly because of neglecting the base currents. \n\n">
</Paintings>

domi8space, 2024-10-25:
> This repository is not related to Qucs-S. The issues and proposals for Qucs-S should be reported here: https://github.com/ra3xdh/qucs_s/issues https://github.com/ra3xdh/qucs_s/discussions https://github.com/ra3xdh/qucsator_rf/issues

Is it a good practice here to duplicate this request, if justified, in Qucs-S' GitHub as well? If yes I can do it. 

> I need more info how this is used for circuit modelling. Usually the resistor connected in series to p-n junction is sufficient and explicit Lambert function is not required. I cannot remember if proprietary circuit simulation software (Spectre, HSPICE) have the support of Lambert function.

As stated above, from my point of view this function would be more useful for postprocessing, however I do not exclude that some users may need it for device modelling.

ra3xdh, 2024-10-25:
> Is it a good practice here to duplicate this request, if justified, in Qucs-S' GitHub as well? 

No duplication is required. I will keep this in mind while preparing the next version of QucsatorRF.