Qucs/qucsator #56: Microstrip T-junction: unphysical results at high frequencies

smaslovski, 2025-05-10:
Hi,

As was initially communicated by @aum on t.me/precisionanalog group, qucksator may produce unphysical results when modeling microstrip T-junctions at frequencies approaching the cutoff frequency for the higher-order modes, although commercial simulators such as ADS may still produce reasonable results at these frequencies.

The following figure illustrates the problem:

[image]

In this example, the unphysical results appear in the frequency region from about 8.5 to 10.2 GHz. It is also seen that in this frequency range the values of S-parameters are NaNs.

Investigation of the T-junction analytical model formulation used in qucsator (https://qucs.sourceforge.net/tech/node81.html) reveals that the source of this problem is their formula (11.216), which has a term $\sqrt{d_a\cdot d_b}$. As can be seen from Eq. (11.210) on the same page, either $d_a$ or $d_b$ can become negative at high enough frequencies, in which case the product $d_a\cdot d_b$ can also become negative. This results in NaNs in Eq. (11.216).

Apparently, Eq. (11.216) is "original art" by the authors of qucs, although the T-junction model itself can be traced back to the article by E. Hammerstad, ``Computer-Aided Design of Microstrip Couplers with Accurate Discontinuity Models,'' Symposium on Microwave Theory and Techniques, pp. 54-56, June 1981. In this article, Hammerstad gives a "receipt" on how to generalize his formula for the shunt susceptance $B_T$ of a symmetric T-junction (Eq. 14) to the same for an nonsymmetric junction (check the highlighted text):

[image]

Note that Hammerstad writes there: "For the side arm reference plane displacement and the shunt susceptance use the _main line impedance equal to the geometric mean of the two actual ones_." Evidently, Eq. (11.216) in qucs resulted from an incorrect interpretation of this paragraph, since, besides taking the geometric mean of the line impedances (Eq. 11.214), the qucs authors decided also to take a geometric mean of the plane displacements $d_a$ and $d_b$, which does not have any physical meaning.

Indeed, it is evident from physical reasoning that under first-order approximation, the expression for $B_T$ must be linear in $d_a$ and $d_b$. In the following comments I will present a corrected formula and also provide a patch to fix this problem in qucs (now my wife needs me and I have to switch my attention to her).

smaslovski, 2025-05-11:
Returning back to writing: As I mentioned above, a physically sound generalization of Eq. (14) from the Hammerstad's article to the case on nonsymmetric T-junction must be, to the first approximation, linear in $d_a$ and $d_b$, as the formula (14) itself suggests. Furthermore, the contributions from these two displacements to $B_T$ must be additive. Thus, to generalize Hammerstad's Eq. (14) for the case of nonsymmetric T-junction, we may note that for the symmetric T-junction with $d_a = d_b$, $D_a = D_b$, $Z_a = Z_b$, etc., we may identically rewrite Eq. (14) as

$\frac{B_T}{Y_2} = 5.5 \cdot \frac{\varepsilon_r+2}{\varepsilon_r}\left[1 + 0.9\cdot\log R + 4.5 \cdot R\cdot Q - 4.4\cdot e^{-1.3\cdot R} -20 \left(\frac{Z_2}{\eta_0}\right)^2 \right]\cdot\frac{1}{2D_2}\cdot\left(\frac{d_a D_a}{n_a^2\lambda_a}+\frac{d_b D_b}{n_b^2\lambda_b}\right),$

where $R = \frac{\sqrt{Z_a Z_b}}{Z_2}$ and $Q = \frac{f^2}{f_{p,a} f_{p,b}}$.

For a symmetric T-junction, the above formula gives numerically identical result for $B_T$ as Hammerstad's Eq. (14). On the other hand, when the lines "a" and "b" are not identical, this formula gives the necessary generalization for the nonsymmetric T-junction case, as it is 1) in line with the "recipe" given by Hammerstad in his original article; 2) to the first approximation, is linear with respect to $d_a$ and $d_b$; and 3) the contributions to $B_T$ from the two lines are in complete agreement with the form of Hammerstad's Eq. (14).

smaslovski, 2025-05-11:
Here is the patch for the file mstee.cpp that implements the necessary change in the formula for $B_T$:

```
diff --git a/src/components/microstrip/mstee.cpp b/src/components/microstrip/mstee.cpp
index 47aa2845..151095ad 100644
--- a/src/components/microstrip/mstee.cpp
+++ b/src/components/microstrip/mstee.cpp
@@ -173,10 +173,8 @@ void mstee::calcPropagation (nr_double_t f) {
   Tb2 = std::max (Tb2, NR_TINY);
 
   // shunt susceptance
-  Bt = 5.5 * std::sqrt (Da * Db / lda / ldb) * (er + 2) / er /
-    Zl2 / std::sqrt (Ta2 * Tb2) * std::sqrt (da * db) / D2 *
-    (1 + 0.9 * std::log (r) + 4.5 * r * q - 4.4 * std::exp (-1.3 * r) -
-     20 * sqr (Zl2 / Z0));
+  Bt = 5.5 * (er + 2) / er * 0.5 * (da * Da / (Ta2 * lda) + db * Db / (Tb2 * ldb)) / (D2 * Zl2) *
+     (1 + 0.9 * std::log (r) + 4.5 * r * q - 4.4 * std::exp (-1.3 * r) - 20 * sqr (Zl2 / Z0));
 }
 
 /* This function can be used to create an extra microstrip circuit.

```

After applying this patch, the example T-junction simulation from above produces the following result:

[image]

 There are no NaNs anymore in the frequency range from 8.5 to 10.2 GHz. Moreover, after such a correction the results in qucs are similar to the results obtained for the same structure in a commercial simulator (ADS):

[image]

Note that ADS uses a closed-source proprietary model for the microstrip T-junction. The precise analytical formulation of their model, as far as I can tell, was never published anywhere. The only information that I have about it, is from the ADS manual:

[image]

ra3xdh, 2025-05-11:
@smaslovski , See https://github.com/ra3xdh/qucsator_rf/pull/39