Qucs/qucsator #57: Qucsator is no longer developped

felix-salfelder, 2025-05-12:
The goal of this repo is to provide an archived, legacy Qucsator reference implementation. The Qucs-S project maintains a Qucsator fork still under development.

Future [Qucs](https://github.com/Qucs/qucs/blob/develop/ROADMAP) will interoperate with Gnucap. Gnucap is a modern and modular circuit analysis tool and simulator with development currently funded by NLnet (EU commission). Qucsator will be obsolete once this work is complete.

In the meantime, [Gnucsator](https://github.com/Qucs/gnucsator]) acts as a (partial) drop in replacement for Qucsator that plays well with the current GUI. It offers a proving ground for new ideas, and a way forward, see [ROADMAP](https://github.com/Qucs/gnucsator/blob/develop/ROADMAP). Advanced users are encouraged to directly use Qucs schematics and/or circuits from other sources in Gnucap.

Gnucsator makes available all devices implemented for Gnucap, including relevant devices from known SPICE implementations, and a growing subset of Verilog-AMS including some RF modelling features. It provides (most) devices from Qucsator, see [STATUS](https://github.com/Qucs/gnucsator/blob/develop/STATUS). Still, only the latter are currently accessible from the Qucs GUI.

ra3xdh, 2025-05-13:
@felix-salfelder You may consider marking this repository as public archive: https://docs.github.com/en/repositories/archiving-a-github-repository/archiving-repositories

felix-salfelder, 2025-05-13:
Thanks Vadim, I have reworded the first sentence for more clarity. An archived repo is read-only and not fit for the purpose.